wagiminator/ATmega-Soldering-Station #7: chinese clone handle compatibility

keiser1080, 2020-10-02:
Hi, 

is your aviator plug wiring compatible with the chinese handle?
It's a bit confusing.
for example + is pin 2 and - pin 5 & 4.
on the ksger clone & other i see + is pin 5 & ground pin 4.
as you can see here [ksger](https://ae01.alicdn.com/kf/HTB1DNYkbBOD3KVjSZFFq6An9pXag.jpg)

wagiminator, 2020-10-02:
Hi,
I'm not sure if your picture is correct. I can't imagine that the tilt switch is connected to earth. However, there is no official standard. I'm using 4-pin handles from china, e.g. https://aliexpress.com/item/32925815648.html, https://aliexpress.com/item/33046441829.html, https://aliexpress.com/item/4000130747937.html, https://aliexpress.com/item/32891866988.html (all tested). If you already have a handle, you can use a multimeter to figure out the correct pinout and wire the aviator plug accordingly.

keiser1080, 2020-10-02:
> Hi,
> I'm not sure if your picture is correct. I can't imagine that the tilt switch is connected to earth. However, there is no official standard. I'm using 4-pin handles from china, e.g. [...]

Yes i have a handel, i have buy the following station []( https://www.aliexpress.com/item/4000183089084.html)
The station come without the tips.
I tryied using my power cord  the screen show error .
Then as soon as i recieved my tips from banggood i have tryed again.
My home fruse go down.
I tryed separatly the 24 power supply.
And the crontroler using my lab power supply then i see a blue smoke :) coming from a the controler.
I have checked the controler and i found 24v on the output of the regulator.
Conclusion the  controler is totaly broken.
So i am planning to do your controller but with a esp32 in place of the atmega.

But i am still not sure what cause the problem.
I found a shortcut on the aviator plug between the pin labeled E & pin labeled G

And now i found the ntc on the handel is also broken (the heatsrink  do not cover entierly & there is a suspicious mark)
My handel is connected like this:
pin 1 : red   (first pin from right of the tip)
pin 2 : withe connected to ntc toward the pin 5 
pin 3 : green  (third pin from right of the tip)
pin 4 : blue connected to the mercury switch toward pin 3
pin 5 : black  (second pin from right of the tip)

pin label on my solder station:
1:+ 
2:N
3:S
4:N
5:G

closed 2020-11-18 by keiser1080