wagiminator/ATmega-Soldering-Station #12: Is KSGER 5 pin handle compatible

ariefadha, 2021-01-30:
Sir, can you give me schematic for the handle wiring to T12 bits? its seems KSGER handle have difference wiring, and it has NTC. but is it you design need NTC?

If the wiring are difference i need schematic so i can rewire the KSGER handle

Here a forum page that show KSGER handle wiring :
https://www.eevblog.com/forum/reviews/stm32-oled-digital-soldering-station-for-t12-handle/75/

Thanks

wagiminator, 2021-01-30:
I suspect they have the same pinout. Nevertheless, to be on the safe side, you should check it with the multimeter. The earth pin should be conductive with the soldering tip. There should be a resistance of around 8 ohms between (+) and (-) pin. Depending on the inclination of the handle, there should be either a conductive or no connection between SW and (-). The resistance between NTC pin and (-) should be well above 8 Ohms. The NTC pin can be left unconnected or can be connected to (-). In the documentation folder there is a tip schematic in the intructions_v1.0.pdf.

JManDoo, 2021-05-21:
[image]

Here is the pinout I used for my KSGER Handles