wagiminator/ATmega-Soldering-Station #13: T12 Thermocouple uV per degree Celsius question

SmallpTsai, 2021-02-06:
Hello, 
In the readme, it said the delta is roughly **40** uV per degree Celsius
but that is no match to the default calibration temp in firmware.
Please help to correct me If my calculation is wrong.

0. Vcc = **5V**
1. the gain of opamp is set to **201** (1 + 200K/1K)
2. default calibration:
   - 200/216 ->   (200 * 5 * 1000000) / (1024*201*216) = 22.49
   - 280/308 ->  (280 * 5 * 1000000) / (1024*201*308) = 22.08
   - 360/390 ->   (360 * 5 * 1000000) / (1024*201*390) = 22.42

so the actual delta should be ~**22**

Is there any calculation error above? 
Please help me to understand the logic.
Thanks !!

wagiminator, 2021-02-06:
You're right, there's a mistake in the description. I will correct that. The 40uV was based on the assumption that there is a k-type thermocouple in the T12 soldering tip. But that is obviously not the case. The corresponding values in the software are derived empirically and have worked so far.

closed 2021-02-06 by wagiminator

SmallpTsai, 2021-02-06:
thanks for your answer!