wagiminator/ATmega-Soldering-Station #14: charge pump question

SmallpTsai, 2021-02-07:
Hello, 

I am studying your "charge pump" design, please correct me if my understandings are wrong.
Thanks a lot!

(based on board version v2.7)
[image]

**Left:**
D9 on -> Q1 on, current go through Q1, so Q2 Vgs is 0, Q2 off.
C7 is charged to full (~24v if ignore D1 voltage drop) 

**Right:**
D9 off -> Q1 off, C7 start to discharge, Q2 Vgs become 24V, Q2 on.
(Vsource = ~23v, Vgate = ~47v)

**part requirement:**
Q1: Vce need to handle > 47v, therefore choose FMMT619
C7: require > 24V voltage
D1: require > 24V reverse voltage (in discharge scenario)
D4: 18V zener is to protect Q2 gate, IRLR7843 max Vgs = 20V

Thanks.

wagiminator, 2021-02-07:
Precisely! The drawing is great, can I use it in the description? Maybe that will help others too.

SmallpTsai, 2021-02-07:
sure, please feel free to use it.
thanks very much for answering and this interesting project !!

Now that everything is clear, I will try to make my own clone soon :)

closed 2021-02-07 by SmallpTsai

wagiminator, 2021-02-07:
Many Thanks! I wish you every success with your version and look forward to hearing from the result.