wagiminator/ATmega-Soldering-Station #17: What is the role of the op amp on the right side of lmv358？

Eddddddddy, 2021-07-19:
I am trying to understand the right op amp. I don’t know why you design it like this. I'm really not good at this.
[image]

wagiminator, 2021-07-19:
Hi, the second OpAmp of the LMV358 is not used. An unused OpAmp should not float, as this can interfere with the OpAmps in use or, in the worst case, lead to its destruction. The two resistors stabilize the second OpAmp by forming a unity gain amplifier (or buffer) with VCC/2 as a fixed positive input value.

Eddddddddy, 2021-07-20:
Thanks for this answer. I got it!

Eddddddddy, 2021-07-20:
by the way. Can I use OPA333 replace LMV358? I rebuild the PCB project. OPA333 is much smaller than LMV358.

wagiminator, 2021-07-21:
Yes, of course, that should work fine.

closed 2021-07-23 by Eddddddddy