wagiminator/ATmega-Soldering-Station #20: Some questions about P+

SMU-Terry, 2021-11-10:
[image]
the P+ net actually directly connects with GND, is that correct? hope to get a reasonable answer, thanks

wagiminator, 2021-11-10:
No, it doesn't. Connections are marked with a junction dot: https://www.autodesk.com/products/eagle/blog/top-10-tips-draw-next-schematic-design-like-pro/

SMU-Terry, 2021-11-12:
thank you so much, I've got it, and I see that compare to the low version, you add D4 as a voltage regulator in Version 2.7, may I ask where is this idea comes from?

wagiminator, 2021-11-13:
I added that to protect the MOSFET. The Zener diode D4 limits the voltage between the gate and source of the MOSFET to about 18V. Since the maximum gate-source voltage of the MOSFET is 20V, but this is exceeded at higher supply voltages of the soldering station, this could lead to damage to the MOSFET in the long term without the diode.

SMU-Terry, 2021-11-14:
thank you so much!

closed 2021-11-14 by SMU-Terry