wagiminator/ATmega-Soldering-Station #23: Soldering_Station v2.7 flashed BUT no Output on Display

suizideFloat, 2022-05-20:
Dear @wagiminator ,

thanks for the great project. 
I managed to build the PCB and solder all components. Flashing process via an Arduino UNO seems successful, but the (new) display stays dark. 

> avrdude: verifying ...
> avrdude: 24006 bytes of flash verified
> 
> avrdude: safemode: Fuses OK (E:00, H:00, L:00)
> 
> avrdude done.  Thank you.

I haven't connected any wires to V,P,E and SW yet, but the display should show something when the board is powered via ICSP right?
Can I test the display in any way?

wagiminator, 2022-11-24:
Yes, the display should show something. Are all connections correct, has the correct display been used, does it work with other programs?

coldfire0101, 2023-02-24:
im having the same problem. have tried several different sketches with arduino ide and hex files with avrdude but nothing happens on the screen, ive also tried different screens. all i get is the led lighting up on the solder station pcb