wagiminator/ATmega-Soldering-Station #26: JBC c210 handle

coldfire0101, 2023-01-07:
is it possible to use a jbc c210 handle and tip, from what i have seen the jbc only has 3 wires

wagiminator, 2023-01-15:
Sorry, I'm not familiar with the c210 tips.

coldfire0101, 2023-01-15:
thanks for responding. this is what i have for c210 tips, whats throwing me is the pic shows sense a on positive and sense b on negative, otherwise its pretty much the same as t12 tip i think
[image]

wagiminator, 2023-01-18:
Looks very similar to the T12 tips, except for the TC, but that's not used anyway. There is obviously no tilt switch. I would say try it out.