wagiminator/ATmega-Soldering-Station #29: PID buzzing noises.

dahayer, 2023-02-18:
I have a strange buzzing noises (not from the buzzer) when using PID controlling, but  everything works like a charm.
I cannot find the responsible component. Any help will be appreciated. Thanks

wagiminator, 2023-02-21:
Strange. Is the noise coming from the soldering station or another device nearby?

dahayer, 2023-02-22:
Its from the soldering station i have recorded the noises with my phone if you want to hear it.
[https://drive.google.com/file/d/1bfbQFNnSnpNc9TIUcPRwxzufM4XHLMXw/view?usp=sharing](https://drive.google.com/file/d/1bfbQFNnSnpNc9TIUcPRwxzufM4XHLMXw/view?usp=sharing)

wagiminator, 2023-02-22:
This could be one of the ceramic capacitors (singing capacitors phenomenon), possibly C7 or C13. You can try using capacitors with the same values but from different manufacturers, or change the PWM frequency in the code.

dahayer, 2023-02-22:
Thank you very much for your help.

closed 2023-02-22 by dahayer