wagiminator/ATmega-Soldering-Station #31: map logic of calibration?

SmallpTsai, 2023-05-30:
Hello:

I found below lines interesting:
https://github.com/wagiminator/ATmega-Soldering-Station/blob/master/software/original/sources/SolderingStation2_u8glib_v1.9.ino#L392

according to arduino.cc, map syntax are
`map(value, fromLow, fromHigh, toLow, toHigh)`

so, should it be
`map (RawTemp,   21, CalTemp[CurrentTip][0], 0, 200);`
?
please correct me If I misunderstand the map logic here.

Thanks.

wagiminator, 2023-05-31:
Hi, the function maps the measured ADC value to the calibrated temperature. For example, CalTemp[0] is the calibrated temperature at a measured ADC value of 200. This means that the value 200 must be mapped exactly to this temperature and not the other way around, which is the case here.

SmallpTsai, 2023-06-01:
hi, 

> CalTemp[0] is the calibrated temperature at a measured ADC value of 200

Ok, I did misunderstand the calibration process.

I thought the process is:
1. During calibration, you have to rotate encoder until tips thermometer read 200 degree (step 1).
2. CalTemp[0] is the measured ADC value when calibrated temperature reach 200.

But the correct process is:
1. During calibration, the station will hold at ADC value 200 (step 1), you have to rotate encoder so the number displayed match tips thermometer's reading.
2. CalTemp[0] is the calibrated temperature at a measured ADC value of 200

Thanks for clarification!

closed 2023-06-01 by SmallpTsai