wagiminator/ATmega-Soldering-Station #32: Uncontrolled heating of the soldering iron head, power indicator flashing

LLC-zh, 2023-08-18:
When I solder the entire board and connect the 24v3a power supply, everything looks normal, but when I insert the soldering iron tip it goes black, the power indicator blinks, and the iron tip heats up. I don't understand why this is, I need help, thank you.

LLC-zh, 2023-08-20:
To add, 12v1a is fine, just slow to heat up, but 24v3a still doesn't work.

wagiminator, 2023-08-20:
I suspect that the power adapter is a bit overwhelmed at 3A and switches off temporarily, which would also explain the blinking of the power LED. You can check my theory with a multimeter: measure the input voltage at the station and see if the voltage fluctuates. If so, try a different power adapter.

LLC-zh, 2023-08-20:
When the soldering station is unloaded I measured 24v on the lr7843, 5v on one end of the 78m05 and 24v on the other, 24v on the vin,no voltage on the P+. When I insert the soldering iron tip all the voltages become unstable.

wagiminator, 2023-08-20:
So I think it's the power adapter. Do you have a lab bench power supply?

LLC-zh, 2023-08-24:
Yes,you are right,I bought a new power adaptor,it worked fine,thank you again.

wagiminator, 2023-08-30:
Great!

closed 2023-09-13 by LLC-zh