wagiminator/ATmega-Soldering-Station #40: Weird crashesh when giving 20v and solder heating up more then 500°c

Rtyuakatsuki, 2025-05-09:
Hallo i have problem about when using 20v power supply or 19v+ 
It pretend weird when give power more then 19v 

How can i fix it ? 
I see your update just pull off (not use the zener 18v)

I kinda hesitant about it 
And i wanna try the 20v power 
I need faster heating 
That why i ask for the solution what i need to do to fix it 

I use board from local youtuber i got 
Fareed read
Here the link board schematic 

https://blogger.googleusercontent.com/img/b/R29vZ2xl/AVvXsEitz7ZCWq4CHpV2WZGww2bgKEEEbje4CBCZLly9iTcXJruOM4NMSV36XI12CHHk9Jenuw0BcbYYxRtse0hr5YZCM6x_k3Vh-BbzjC0wccpHDqx9-WM4c4vuHTEzO_zFS8a5FvDoloVhuxW6UzK4Hva-xYl9dvyhrzmeJmUU9PbRo4H2ezq0gMT8AS17GA/s1750/ATmegaSolderingStation%20l.jpg

wagiminator, 2025-05-11:
Have you tried using a different 20V power supply? The Zener diode is only there to protect the MOSFET.

Rtyuakatsuki, 2025-05-11:
I tries 2 power supply 
First 19v 60 watt laptop power supply from asus 
And second from lenovo 20v powersupply with power 240watt 

Yes got same problem 
After some minute it weird glitch came out and messed up 
The value default temprature or adc reading 

And i check the adc reading also weird 
When i turn on the board adc reading was started with 
250+ value 

When i plug cheap 12v 3a power supply it was running good 

And after couple hour after i posted that 
I check the ic pin d10 sleep mode 
It shorted to ground 
So i was ordering alternative 
Atmega328p 
LGT8F328p 

Hope can work on it 

So the main question 
What make it happen ? 

And more detail 
I tried search about ADC problem 
With power supply 
It cause by noise from the power supply 

But it kinda weird 
Expensive server power supply 
Have bad noise isolation
And the cheap 12v 3a can handle well 

So i was rethink again 
Its not from power supply i think there something wrong in circuit but i dont know where 

Fyi 
I already ordered 
Ss54 
And fmmt617(i dont know it would work or not but i just wanna try because in local supplier i live 
I cannot found it

Rtyuakatsuki, 2025-05-19:
Update 

After changhing 
Transistor to fmmt617 
And diode schottky ss54

Replace atmega328p ic 
I have 3 ic brand new ic 

Got weird on off everytime pid used 

Only pid 

Direct control 
Nothing wrong with it